mkolender/rtl8821CU #3: Failed to compile on Raspberry Pi 3

leonzhang1109, 2018-06-25:
The error I got during compilation is:
`gcc: error: unrecognized command line option ‘-mgeneral-regs-only’`

The whole log is:
`pi@raspberrypi:~/rtl8821CU $ make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.9.35-v7+/build M=/home/pi/rtl8821CU  modules
make[1]: Entering directory '/home/pi/linux-be2540e540f5442d7b372208787fb64100af0c54'
  CC [M]  /home/pi/rtl8821CU/core/rtw_cmd.o
gcc: error: unrecognized command line option ‘-mgeneral-regs-only’
scripts/Makefile.build:293: recipe for target '/home/pi/rtl8821CU/core/rtw_cmd.o' failed
make[2]: *** [/home/pi/rtl8821CU/core/rtw_cmd.o] Error 1
Makefile:1490: recipe for target '_module_/home/pi/rtl8821CU' failed
make[1]: *** [_module_/home/pi/rtl8821CU] Error 2
make[1]: Leaving directory '/home/pi/linux-be2540e540f5442d7b372208787fb64100af0c54'
Makefile:1918: recipe for target 'modules' failed
make: *** [modules] Error 2
pi@raspberrypi:~/rtl8821CU $ 
`

whitebatman2, 2018-06-27:
I don't own Raspberry Pi 3, but from what I've read on internet the problem is not with the module, but you are using wrong version of gcc. With this version you wouldn't probably be able to build any kernel module for your kernel at all. What OS are you using and which version of gcc?

DrSchottky, 2018-07-12:
Compile it for RPI and it'll work on RPI3 too. Raspbian and Pi3 (by default, can be changed) are ARMv7

closed 2018-08-15 by whitebatman2