mkolender/rtl8821CU #5: 4.15.17 kernel backtrace with device 0bda:c811

slavkoja, 2018-08-22:
I recently buy toy from ebay, which is identified by USB ID:
```
ID 0bda:c811 Realtek Semiconductor Corp.
```
I compiled your driver on 4.15.17 kernel  with some warnings, then i install it manually:
```cp tmp/rtl8821CU/8821cu.ko /lib/modules/4.15.17-1/extra/
depmod $(uname -r)
```
When i plug device, i got backtrace:
```
Aug 22 08:46:23 bonifac kernel: [ 5616.959173] usbcore: registered new interface driver rtl8821cu
Aug 22 08:46:23 bonifac kernel: [ 5616.976864] rtl8821cu 2-1.4:1.0 wlan121: renamed from wlan0
Aug 22 08:46:23 bonifac kernel: [ 5617.002509] rtl8821cu 2-1.4:1.0 wlan1: renamed from wlan121
Aug 22 08:46:23 bonifac kernel: [ 5617.026380] ------------[ cut here ]------------
Aug 22 08:46:23 bonifac kernel: [ 5617.026382] name 'wlan0'
Aug 22 08:46:23 bonifac kernel: [ 5617.026412] WARNING: CPU: 2 PID: 1800 at /var/tmp/portage/sys-kernel/debian-sources-4.15.17-r1/work/linux-4.15.17/fs/proc/
generic.c:561 remove_proc_entry+0xff/0x140
Aug 22 08:46:23 bonifac kernel: [ 5617.026413] Modules linked in: 8821cu(O) cfg80211 rfkill nls_utf8 isofs cbc cts rpcsec_gss_krb5 auth_rpcgss nfsv4 dns_reso
lver nfs lockd grace fscache sunrpc iptable_nat nf_nat_ipv4 nf_nat nf_log_ipv4 nf_conntrack_ipv4 nf_defrag_ipv4 iptable_filter ip_tables nf_log_ipv6 nf_log_c
ommon xt_LOG xt_limit xt_comment xt_tcpudp nf_conntrack_ipv6 nf_defrag_ipv6 xt_multiport xt_conntrack nf_conntrack ip6table_filter ip6_tables x_tables binfmt
_misc reiserfs dm_mod br_netfilter bridge stp llc fuse tty0tty(O) tun loop pci_stub vboxpci(O) vboxnetadp(O) vboxnetflt(O) vboxdrv(O) it87 hwmon_vid usblp jo
ydev hid_generic usbhid hid snd_hda_codec_hdmi cp210x ch341 usbserial btrfs zstd_compress zstd_decompress xxhash xor pktcdvd raid6_pq nvidia_drm(PO) nvidia_m
odeset(PO) edac_mce_amd kvm_amd evdev nvidia(PO) kvm irqbypass
Aug 22 08:46:23 bonifac kernel: [ 5617.026494]  crct10dif_pclmul snd_hda_codec_realtek crc32_pclmul crc32c_intel ohci_pci snd_hda_codec_generic ghash_clmulni
_intel pcbc aesni_intel snd_hda_intel aes_x86_64 snd_hda_codec crypto_simd efi_pstore glue_helper drm_kms_helper cryptd snd_hda_core snd_hwdep pcspkr drm snd
_pcm xhci_pci r8169 snd_timer efivars ipmi_devintf ohci_hcd ehci_pci k10temp fam15h_power ipmi_msghandler mii xhci_hcd ehci_hcd snd soundcore sp5100_tco usbc
ore shpchp usb_common i2c_piix4 button acpi_cpufreq efivarfs initio BusLogic arcmsr aic7xxx aic79xx scsi_transport_spi sr_mod cdrom sg sd_mod pdc_adma sata_m
v ata_piix ahci libahci sata_qstor sata_vsc sata_uli sata_sis pata_sis sata_sx4 sata_nv sata_via sata_svw sata_sil24 sata_sil sata_promise libata scsi_mod ex
t4 fscrypto jbd2 mbcache crc16 [last unloaded: 8821cu]
Aug 22 08:46:23 bonifac kernel: [ 5617.026574] CPU: 2 PID: 1800 Comm: udevd Tainted: P        W  O     4.15.17-1 #1
Aug 22 08:46:23 bonifac kernel: [ 5617.026575] Hardware name: Gigabyte Technology Co., Ltd. To be filled by O.E.M./970A-DS3P, BIOS FB 10/31/2014
Aug 22 08:46:23 bonifac kernel: [ 5617.026580] RIP: 0010:remove_proc_entry+0xff/0x140
Aug 22 08:46:23 bonifac kernel: [ 5617.026582] RSP: 0018:ffffbae7442afc40 EFLAGS: 00010286
Aug 22 08:46:23 bonifac kernel: [ 5617.026585] RAX: 0000000000000000 RBX: ffffbae740d0b8f8 RCX: 0000000000000006
Aug 22 08:46:23 bonifac kernel: [ 5617.026587] RDX: 0000000000000007 RSI: 0000000000000086 RDI: ffff99ce2ec966f0
Aug 22 08:46:23 bonifac kernel: [ 5617.026589] RBP: 0000000000000000 R08: 000000000000054e R09: 0000000000000001
Aug 22 08:46:23 bonifac kernel: [ 5617.026591] R10: ffffbae7442afb60 R11: 0000000000000001 R12: ffff99cd449523c0
Aug 22 08:46:23 bonifac kernel: [ 5617.026593] R13: ffffbae740d07000 R14: ffff99cd5e0ee000 R15: ffff99cd449523c0
Aug 22 08:46:23 bonifac kernel: [ 5617.026596] FS:  00007f4e88cdb2c0(0000) GS:ffff99ce2ec80000(0000) knlGS:0000000000000000
Aug 22 08:46:23 bonifac kernel: [ 5617.026598] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
Aug 22 08:46:23 bonifac kernel: [ 5617.026600] CR2: 00007f834c06a7bc CR3: 00000001c1722000 CR4: 00000000000406e0
Aug 22 08:46:23 bonifac kernel: [ 5617.026602] Call Trace:
Aug 22 08:46:23 bonifac kernel: [ 5617.026700]  rtw_adapter_proc_replace+0x65/0x87 [8821cu]
Aug 22 08:46:23 bonifac kernel: [ 5617.026776]  rtw_ndev_notifier_call+0x3b/0x3d [8821cu]
Aug 22 08:46:23 bonifac kernel: [ 5617.026783]  notifier_call_chain+0x47/0x70
Aug 22 08:46:23 bonifac kernel: [ 5617.026788]  dev_change_name+0x1c5/0x2e0
Aug 22 08:46:23 bonifac kernel: [ 5617.026794]  dev_ifsioc+0x211/0x420
Aug 22 08:46:23 bonifac kernel: [ 5617.026800]  ? security_capable+0x47/0x60
Aug 22 08:46:23 bonifac kernel: [ 5617.026804]  dev_ioctl+0x30a/0x580
Aug 22 08:46:23 bonifac kernel: [ 5617.026808]  ? sk_prot_alloc+0x2f/0x120
Aug 22 08:46:23 bonifac kernel: [ 5617.026813]  ? sock_do_ioctl+0x41/0x50
Aug 22 08:46:23 bonifac kernel: [ 5617.026817]  sock_do_ioctl+0x41/0x50
Aug 22 08:46:23 bonifac kernel: [ 5617.026821]  sock_ioctl+0x19a/0x2b0
Aug 22 08:46:23 bonifac kernel: [ 5617.026825]  do_vfs_ioctl+0xa4/0x630
Aug 22 08:46:23 bonifac kernel: [ 5617.026830]  ? sock_alloc_file+0xa4/0x150
Aug 22 08:46:23 bonifac kernel: [ 5617.026833]  SyS_ioctl+0x74/0x80
Aug 22 08:46:23 bonifac kernel: [ 5617.026838]  do_syscall_64+0x6e/0x130
Aug 22 08:46:23 bonifac kernel: [ 5617.026843]  entry_SYSCALL_64_after_hwframe+0x3d/0xa2
Aug 22 08:46:23 bonifac kernel: [ 5617.026847] RIP: 0033:0x7f4e8837dd87
Aug 22 08:46:23 bonifac kernel: [ 5617.026849] RSP: 002b:00007ffd5c7d5988 EFLAGS: 00000246 ORIG_RAX: 0000000000000010
Aug 22 08:46:23 bonifac kernel: [ 5617.026852] RAX: ffffffffffffffda RBX: 0000000000000007 RCX: 00007f4e8837dd87
Aug 22 08:46:23 bonifac kernel: [ 5617.026854] RDX: 00007ffd5c7d59b0 RSI: 0000000000008923 RDI: 0000000000000007
Aug 22 08:46:23 bonifac kernel: [ 5617.026855] RBP: 00007ffd5c7d59b0 R08: 0000000000000003 R09: 00007f4e883cabd0
Aug 22 08:46:23 bonifac kernel: [ 5617.026857] R10: 0000000000000559 R11: 0000000000000246 R12: 00007ffd5c7d59c0
Aug 22 08:46:23 bonifac kernel: [ 5617.026859] R13: 00007ffd5c7d5a40 R14: 0000000001c7eac0 R15: 0000000001c7eac0
Aug 22 08:46:23 bonifac kernel: [ 5617.026861] Code: 83 f8 48 75 2d 48 89 ef e8 cf fc ff ff e9 6a ff ff ff 48 8b 04 24 83 68 08 01 eb d3 48 89 de 48 c7 c7 65 7b e6 90 e8 91 ce dc ff <0f> 0b e9 4a ff ff ff 48 8b 75 30 48 8d 8d 8d 00 00 00 4c 8d 40 
Aug 22 08:46:23 bonifac kernel: [ 5617.026922] ---[ end trace f9bdf744d182e48d ]---
```

traud, 2018-08-25:
No answer to your report but a question. I am looking for IEEE 802.11ac Wave 2 with 1×1:1, the [RTL8811CU](https://wikidevi.com/wiki/Realtek#ac). And I want one with generic/default IDs. Which Wi-Fi USB adapter did you buy exactly:

- D-Link DWA-171 (Hardware Version C1),
- Tenda U9,
- TP-Link TL-WDN5200H,
- Totolink A650UA, or
- Zioncom A650UA?

slavkoja, 2018-08-25:
Noname from ebay https://www.ebay.com/itm/USB-Dual-band-2-4G-5G-Wireless-Adapter-600Mbps-Wifi-Receiver-Dongles-RTL8811AU/183198742133

whitebatman2, 2018-08-26:
This driver is compatible only with rtl8821CU and rtl8811CU AFAIK. Even though VEN and DEV ids are identical and names are very similar it just won't work. Please try this driver: https://github.com/diederikdehaas/rtl8812AU

slavkoja, 2018-08-26:
I tried mentioned driver as first, but the device was not recognised (no wlanX nor similar device created), then i search by ID and i found your driver. It is terrible to identify chip inside, because ebay's descriptions are not reliable at all and the "original" package doesn't mention used chip...

Anyway, thanks for reply ;-)

traud, 2018-08-27:
The easiest way would be to open the case and double-check the printings on the chip. Anyway, do you have any FCC ID printed on the case of adapter?

You can test whether that other driver was loaded/matched via `lsusb --tree`. As of today, the latest driver for a RTL8811**A**U can be found here: https://github.com/aircrack-ng/rtl8812au/tree/v5.3.4
Before you go for `make`, you have to change the file `os_dep/linux/usb_intf.c`: Search for `0x0811`and change that to `0xc811`. Of course, you have to `sudo make uninstall` the two previous drivers first. Then, this one should at least load/match. If you get a backtrace again, you did everything right.

slavkoja, 2018-08-28:
Thanks, i will try it, but please, i don't understand your latest sentense - the backtrace is expected?

traud, 2018-08-29:
I hope you do not get a backtrace and everything works. Nevertheless, if you get a backtrace with that driver as well, you know that at least loading/matching worked.

slavkoja, 2018-08-29:
OK, i understand now, thanks.

Anyway, you can close it now or leave as note about non-conforming USB ID... I will leave decision to you ;-)

traud, 2018-08-30:
@slavkoja were you able to use that Wi-Fi adapter with that other driver (after patching it)? If not either, I guess it is time to crack the housing/plastics and look inside which chip is used actually. Keep us updated! Especially, when you find other identification marks like a FCC ID or a manufacturer/product name.

slavkoja, 2018-09-01:
@traud you asked for result. Not very good. I compiled that version of driver, device seems to be recognised by it, but then USB ID changes to ``0bda:1a2b`` and i have no wireless devices, see:
```
new high-speed USB device number 7 using xhci_hcd
New USB device found, idVendor=0bda, idProduct=c811
New USB device strings: Mfr=1, Product=2, SerialNumber=3
Product: 802.11ac NIC
Manufacturer: Realtek
SerialNumber: 123456
reset high-speed USB device number 7 using xhci_hcd
device firmware changed
USB disconnect, device number 7
new high-speed USB device number 8 using xhci_hcd
config 1 interface 0 altsetting 0 endpoint 0x8A has an invalid bInterval 0, changing to 7
config 1 interface 0 altsetting 0 endpoint 0xB has an invalid bInterval 0, changing to 7
New USB device found, idVendor=0bda, idProduct=1a2b
New USB device strings: Mfr=1, Product=2, SerialNumber=0
Product: DISK
Manufacturer: Realtek
```
Then i crack its case and i fount, that it really has RTL8811**C**U chip (another info on chip: H7P95G1GH45). Then we are back and IMO the issue have to be reopened.

slavkoja, 2018-09-02:
IMO, it doesn't matter if bought from aliexpress or ebay, they are bot the source of uncertainty, which is related to mentioned reliability of their descriptions ;-)

After new kernel arrives into my Funtoo, i will report result again (if i didn't damage it by opening). Current 4.15 kernel is trying to solve spectra/meltdown problems and its results are sometime unpredictably, then the driver can be affected by this...

nvcnvn, 2018-10-11:
Not sure if I should comment on the same issue, I have not make my Totolink A650UA to work with Ubuntu 18.04 with 4.18 kernel.
I have success install your driver using make steps on readme.
```
uname -a
Linux ubuntu 4.18.0-041800-generic #201808122131 SMP Sun Aug 12 21:33:20 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
```

```
lsmod | grep 8821cu
8821cu               1843200  0
cfg80211              663552  3 8821cu,mt7601u,mac80211
```

but the adapter still recognized as a usb storage:
```
lsusb
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 003 Device 006: ID 0bda:1a2b Realtek Semiconductor Corp. 
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 004: ID 148f:7601 Ralink Technology, Corp. MT7601U Wireless Adapter
Bus 001 Device 003: ID 12cf:0285 DEXIN 
Bus 001 Device 002: ID 2516:0020  
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub

lsusb --tree
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 3: Dev 6, If 0, Class=Mass Storage, Driver=usb-storage, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 5: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 4, If 0, Class=Vendor Specific Class, Driver=mt7601u, 480M
```

I try to reject it, but still not work:
```
sudo usb_modeswitch -KW -v 0bda -p 1a2b
Take all parameters from the command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.5.2 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0x1a2b

StandardEject=1

Look for default devices ...
  found USB ID 1d6b:0003
  found USB ID 0bda:1a2b
   vendor ID matched
   product ID matched
  found USB ID 1d6b:0002
  found USB ID 1d6b:0003
  found USB ID 148f:7601
  found USB ID 12cf:0285
  found USB ID 2516:0020
  found USB ID 1d6b:0002
 Found devices in default mode (1)
Access device 008 on bus 003
Get the current device configuration ...
Current configuration number is 1
Use interface number 0
 with class 8
Use endpoints 0x0b (out) and 0x8a (in)

USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
 OK, driver detached
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 1 (CSW) ...
 Response successfully read (13 bytes), status 1
Trying to send message 2 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 2 (CSW) ...
 Response successfully read (13 bytes), status 0
Trying to send message 3 to endpoint 0x0b ...
 Sending the message returned error -1. Try to continue
Read the response to message 3 (CSW) ...
 Response reading failed (error -1)
 Device is gone, skip any further commands
-> Run lsusb to note any changes. Bye!

lsusb --tree
/:  Bus 04.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 480M
    |__ Port 3: Dev 9, If 0, Class=Vendor Specific Class, Driver=, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/8p, 10000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/14p, 480M
    |__ Port 5: Dev 2, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 2, If 2, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 5: Dev 2, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 6: Dev 3, If 1, Class=Human Interface Device, Driver=usbhid, 12M
    |__ Port 7: Dev 4, If 0, Class=Vendor Specific Class, Driver=mt7601u, 480M
```