mkolender/rtl8821CU #8: Compile Error on Raspbian

DanteLee, 2018-10-18:
Hardware: RPi3 B+
OS: Raspbian release on Octo, 9, 2018
Kernel: 4.14.71-v7+
Header Package: apt install raspberrypi-kernel-headers
detail:
 i decide to use dkms to build the module and followed the instructions in README. But i got a Compilation Error which inform that a flag called '-mgeneral-regs-only' can not be recognized. I found a [document](http://infocenter.arm.com/help/index.jsp?topic=/com.arm.doc.dui0774a/chr1392305446662.html) seems to related to this error.
Fortunately, when i change the platform option to rpi (not rpi3), the compilation works. Right now i'm looking for the deep reason and fix for this situation, any idea?

EinSoldiatGott, 2018-10-25:
Dante, I was about to buy a dongle with this chipset to be used with my RBpi3B+, I have a question for you and not an answer to your question sorry.

Is your RPi3 B+ able to recognize the Bluetooth functionality of the dongle? or neither Wifi neither Bluetooth?

Thanks

DanteLee, 2018-10-29:
> Dante, I was about to buy a dongle with this chipset to be used with my RBpi3B+, I have a question for you and not an answer to your question sorry.
> 
> Is your RPi3 B+ able to recognize the Bluetooth functionality of the dongle? or neither Wifi neither Bluetooth?
> 
> does "sudo hciconfig" list this device as available?
> 
> Thanks

My situation is to compile driver for Tenda u9 dongle, a wifi dongle which use rtl8811CU chipset. The problem is after install the driver, linux kernel recognized the device to a mass storage device other than a rtl8821CU wifi deivces which it supposed to be. I guess it's caused by the u9's preinstalled driver in the device( it has a small storage space inside the deivce ).

If your problem is similar to that, add a line in  `os_dep/linux/usb_intf.c` may help. But you should make sure this driver also works for your chipset. After plug the deivce into your rpi, you can use `dmesg` to see wether the device is recognized correctly( By ProdId and VendorID ).

I solved this problem by change my dongle to Tenda U12 with rtl8821AU chipset. [this driver](https://github.com/abperiasamy/rtl8812AU_8821AU_linux) can work perfectly with this chipset.

hope these information helps you.

minscof, 2018-11-24:
I think, I have similar troubles. I have [this wifi/bluetooth usb dongle](https://www.aliexpress.com/item/Comfast-CF-WU910A-Dual-Band-5-8G-USB-WiFi-Adapter-600Mbps-Wifi-Receiver-Wireless-Network-Card/32841531895.html). It has a rtl8821CU chipset inside. 
I compile this driver on my raspberry pi 2 with stretch raspbian `Linux raspberrypi 4.14.79+ #1159,` but the dongle is recognized at boot as a mass storage. I need to run this command
`sudo usb_modeswitch -KW -v 0bda -p 1a2b`
to get the dongle recognized as a wifi dongle and then with lsub command, I get
`Bus 001 Device 005: ID 0bda:c820 Realtek Semiconductor Corp`
But when I reboot the raspberry, I lost this configuration and I must run the above command again.
Is it a way to make the dongle correctly recognized permanently ?

My second trouble is for bluetooth : how can I get it working ?
hciconfig hci0 returns this
`hci0:   Type: Primary  Bus: USB
        BD Address: 00:00:00:00:00:00  ACL MTU: 0:0  SCO MTU: 0:0
        DOWN
        RX bytes:504 acl:0 sco:0 events:71 errors:0
        TX bytes:17714 acl:0 sco:0 commands:72 errors:0`
and when I try hciconfig hci0 up, I get
`Can't init device hci0: Cannot assign requested address (99)`

Can you help me ?

EinSoldiatGott, 2018-11-24:
Hi, thanks for providing the model from Ali, this is exactly the module that I was going to buy because it is one of the few with BT 4.2 with an antenna. I'll wait until it because.

Currently, I'm using one from ZAPO manufacturer, it's BT 4.1 and it works correctly on Raspbian.