mkolender/rtl8821CU #11: Is Bluetooth working?

EinSoldiatGott, 2018-12-21:
Hi, is Bluetooth working for the 8821cu chipset?

Bsplesk, 2019-04-07:
**works** _(Linux 4.19.0-0.bpo.2-amd64 #1 SMP Debian 4.19.16-1~bpo9+1 (2019-02-07) x86_64 GNU/Linux)_. 
example:
> @NAS:~$ bluetoothctl 
> [NEW] Controller XX:XX:XX:XX:XX:XX NAS [default]
> [NEW] Device XX:XX:XX:XX:XX:XX MDR-1000X 
> 
> show
> Controller 54:C9:E0:00:01:98
> 	Name: NAS
> 	Alias: NAS
> 	Class: 0x0c0104
> 	Powered: yes
> 	Discoverable: no
> 	Pairable: yes
> 	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
> 	UUID: Generic At# showtribute Profile (00001801-0000-1000-8000-00805f9b34fb)
> 	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
> 	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
> 	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
> 	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
> 	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
> 	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
> 	Modalias: usb:v1D6Bp0246d052B
> 	Discovering: no

MrCapone, 2019-05-02:
rtl8821c supported since 4.18 kernel. For older kernels use dkms module https://github.com/MrCapone/btrtl-dkms