mkolender/rtl8821CU #13: Change the build instructions

DocSneider, 2019-01-01:
Maybe it would be ok to change the build-instructions  like this:

1. DRV_NAME=rtl8821CU
2. DRV_VERSION=5.2.5.3
3. sudo mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
4. cd /tmp/
5. git clone https://github.com/whitebatman2/rtl8821CU.git
6. cd rtl8821CU/
7. git archive master | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
8. sudo dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
9. sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
10. sudo dkms install -m ${DRV_NAME} -v ${DRV_VERSION}

Because some users are not so familiar with git.

PegasusNZ, 2019-01-08:
Could you also add a comment in the instructions about the Net Manager bug and the fix
echo -e "\n[device]\nwifi.scan-rand-mac-address=no" | sudo tee -a /etc/NetworkManager/NetworkManager.conf

squalldna, 2019-12-01:
I need help on this please.
Im always getting this error. About kernel header.

> pi@raspberrypi:~ $ cd rtl8821CU/
> -bash: cd: rtl8821CU/: No such file or directory
> pi@raspberrypi:~ $ cd /tmp/                                                  pi@raspberrypi:/tmp $ cd rtl8821CU/
> pi@raspberrypi:/tmp/rtl8821CU $ git archive master | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
> pi@raspberrypi:/tmp/rtl8821CU $ sudo dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
> 
> Creating symlink /var/lib/dkms/rtl8821CU/5.2.5.3/source ->
>                  /usr/src/rtl8821CU-5.2.5.3
> 
> DKMS: add completed.
> pi@raspberrypi:/tmp/rtl8821CU $ sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
> Error! echo
> Your kernel headers for kernel 4.9.41-v7+ cannot be found at
> /lib/modules/4.9.41-v7+/build or /lib/modules/4.9.41-v7+/source.

squalldna, 2019-12-02:
Ok so now im stuck here:

> pi@raspberrypi:/tmp/rtl8821CU $ sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}

Kernel preparation unnecessary for this kernel.  Skipping...

Building module:
cleaning build area...
'make' KVER=4.19.75-v7+....(bad exit status: 2)
Error! Bad return status for module build on kernel: 4.19.75-v7+ (armv7l)
Consult /var/lib/dkms/rtl8821CU/5.2.5.3/build/make.log for more information.