mkolender/rtl8821CU #16: Difficulties when compiling on Ubuntu 18.04

such3r, 2019-01-10:
The same issue like in https://github.com/dtaht/sch_cake/issues/110

1. Kernel does not support PIC mode: Added EXTRA_CFLAGS += $(CFLAGS_EXTRA) -fno-pie
2. There is no stdarg.h: Added EXTRA_CFLAGS += $(CFLAGS_EXTRA) -fno-pie -I/usr/lib/gcc/x86_64-linux-gnu/7/include/
3. There is no asm/linkage.h: ln -s /usr/src/linux-headers-4.15.0-29-generic/include/asm-generic /usr/src/linux-headers-4.15.0-29-generic/include/asm
4. There is no asm/thread_info.h: have not found any workaround yet

Linux 4.15.0-29-generic #31-Ubuntu SMP Tue Jul 17 15:39:52 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux
gcc (Ubuntu 7.3.0-27ubuntu1~18.04) 7.3.0

`dpkg -l "*headers*"`
[headers.txt](https://github.com/whitebatman2/rtl8821CU/files/2743684/headers.txt)

marconius, 2019-04-07:
I had a similar issue. It went away for me with most recent changes (0278eaa9937a7c60f2916b5d334e40dfb7870bb0)

nampq309, 2019-11-01:
My kernel is  4.15.0-66-generic. I compiled & install without any error but my PC did not recognize my USB Wifi.
Note: My device is Totolink A650UA, I found its chipset is RTL8811CU.

Please help. Thanks!