mkolender/rtl8821CU #17: can't be compiled in linux 5.0

xfor19120101, 2019-01-19:
linux kernel 5.0 without vla, so, can't compile, source files have many vla, so driver needs to be rewrite to remove vla.

xfor19120101, 2019-01-19:
/* fix for kernel>= 5.0
 *1. and -wno-vla in Makefile, 
 *2. modified os_dep/linux/rtw_android.c
*/
                      goto exit;
	}

#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
	if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
#else
	if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
#endif
		RTW_INFO("%s: failed to access memory\n", __FUNCTION__);
		ret = -EFAULT;
		goto exit;

vhervieux, 2019-05-14:
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index 80a3c01..f5a2b6a 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -334,7 +334,12 @@ static const struct ieee80211_txrx_stypes
 
 static u64 rtw_get_systime_us(void)
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
+/* TODO: fix version in which get_monotic_bootime has disappeared */
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
+       struct timespec64 ts;
+       ktime_get_boottime_ts64(&ts);
+       return ((u64)ts.tv_sec * 1000000) + ts.tv_nsec / 1000;
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
        struct timespec ts;
        get_monotonic_boottime(&ts);
        return ((u64)ts.tv_sec * 1000000) + ts.tv_nsec / 1000;
diff --git a/os_dep/linux/rtw_android.c b/os_dep/linux/rtw_android.c
index 9c85dfb..5bcf6e9 100644
--- a/os_dep/linux/rtw_android.c
+++ b/os_dep/linux/rtw_android.c
@@ -625,8 +625,11 @@ int rtw_android_priv_cmd(struct net_device *net, struct ifreq *ifr, int cmd)
                ret = -ENOMEM;
                goto exit;
        }
-
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
+       if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
+#else
        if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
+#endif
                RTW_INFO("%s: failed to access memory\n", __FUNCTION__);
                ret = -EFAULT;
                goto exit;

vhervieux, 2019-05-14:
[patch](https://pastebin.com/8KZ6qDQn)

zjiekai, 2019-10-08:
```
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index 80a3c01..f5a2b6a 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -334,7 +334,12 @@ static const struct ieee80211_txrx_stypes
 
 static u64 rtw_get_systime_us(void)
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
+/* TODO: fix version in which get_monotic_bootime has disappeared */
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(4, 20, 0))
+       struct timespec64 ts;
+       ktime_get_boottime_ts64(&ts);
+       return ((u64)ts.tv_sec * 1000000) + ts.tv_nsec / 1000;
+#elif (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
        struct timespec ts;
        get_monotonic_boottime(&ts);
        return ((u64)ts.tv_sec * 1000000) + ts.tv_nsec / 1000;
diff --git a/os_dep/linux/rtw_android.c b/os_dep/linux/rtw_android.c
index 9c85dfb..5bcf6e9 100644
--- a/os_dep/linux/rtw_android.c
+++ b/os_dep/linux/rtw_android.c
@@ -625,8 +625,11 @@ int rtw_android_priv_cmd(struct net_device *net, struct ifreq *ifr, int cmd)
                ret = -ENOMEM;
                goto exit;
        }
-
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(5, 0, 0))
+       if (!access_ok(priv_cmd.buf, priv_cmd.total_len)) {
+#else
        if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
+#endif
                RTW_INFO("%s: failed to access memory\n", __FUNCTION__);
                ret = -EFAULT;
                goto exit;
```

```
diff --git a/os_dep/linux/ioctl_cfg80211.c b/os_dep/linux/ioctl_cfg80211.c
index 80a3c01..f5a2b6a 100644
--- a/os_dep/linux/ioctl_cfg80211.c
+++ b/os_dep/linux/ioctl_cfg80211.c
@@ -334,7 +334,12 @@ static const struct ieee80211_txrx_stypes
 
 static u64 rtw_get_systime_us(void)
 {
-#if (LINUX_VERSION_CODE >= KERNEL_VERSION(2, 6, 39))
```

copied from @vhervieux 's patch

luckylittle, 2020-12-13:
@whitebatman2  @zjiekai @xfor19120101 I am unable to compile the latest `master` - can you help, please? Below are the details:
* Red Hat Enterprise Linux release `8.3` (Ootpa)
* Kernel `4.18.0-240.1.1.el8_3.x86_64`
* Error:

```text
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.18.0-240.1.1.el8_3.x86_64/build M=/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2  modules
make[1]: Entering directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
  CC [M]  /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.o
In file included from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/include/drv_types.h:35,
                 from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c:25:
/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/include/osdep_service_linux.h:86,
                 from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/include/osdep_service.h:47,
                 from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/include/drv_types.h:32,
                 from /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c:25:
./include/linux/ieee80211.h:1463: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c: In function ‘rtw_android_priv_cmd’:
/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c:632:62: error: macro "access_ok" passed 3 arguments, but takes just 2
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
                                                              ^
/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c:632:7: error: ‘access_ok’ undeclared (first use in this function)
  if (!access_ok(VERIFY_READ, priv_cmd.buf, priv_cmd.total_len)) {
       ^~~~~~~~~
/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.c:632:7: note: each undeclared identifier is reported only once for each function it appears in
make[2]: *** [scripts/Makefile.build:316: /home/USER/Downloads/WiFi/rtl8821CU-whitebatman2/os_dep/linux/rtw_android.o] Error 1
make[1]: *** [Makefile:1544: _module_/home/USER/Downloads/WiFi/rtl8821CU-whitebatman2] Error 2
make[1]: Leaving directory '/usr/src/kernels/4.18.0-240.1.1.el8_3.x86_64'
make: *** [Makefile:1923: modules] Error 2
```

The above patch by @vhervieux is no longer relevant.