mkolender/rtl8821CU #20: can't work on TI's AM335x platform

godspy16888, 2019-02-01:
i tried on the beagleboneBlack/blue and the pocketbeagle which based on TI's AM335x ...
doesn't work there...maybe cause in its makefile there is no supporting infor for TI's AM335x/ARM??

RobertCNelson, 2019-02-01:
@godspy16888 use the Raspberry Pi option:

```code
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = y
CONFIG_PLATFORM_ARM_RPI3 = n
```

PS, this 'module' was recently added (about a month ago) to the repos included in beagleboard.org's default debian (and ubuntu) images.

```code
sudo apt update
sudo apt install rtl8821cu-modules-`uname -r`
```

Regards,

godspy16888, 2019-02-02:
@RobertCNelson 
Thank you for your quick response!
I've tried but looks BB's latest kernal doesn't include that module.
look the error i came up:

E: Unable to locate package rtl8821cu-modules-4.14.71-ti-r80
E: Couldn't find any package by glob 'rtl8821cu-modules-4.14.71-ti-r80'
E: Couldn't find any package by regex 'rtl8821cu-modules-4.14.71-ti-r80'

but I'm continuing try to clone the whole depositories of BB and place the rtl8821cu's driver into it.
then edit some config files and try again...rtl8821cu should be easily implemented...but i'm not sure can i get it down or not :-)

RobertCNelson, 2019-02-02:
@godspy16888 that kernel is too old it wasn't till r86 in the 4.14.x feed:

http://repos.rcn-ee.net/debian/pool/main/r/

Just run:

```code
cd /opt/scripts/tools/
git pull
sudo ./update_kernel.sh
sudo reboot
sudo apt install rtl8821cu-modules-`uname -r`
sudo reboot
```

Regards,

godspy16888, 2019-02-02:
> @godspy16888 that kernel is too old it wasn't built till r86 in the 4.14.x feed:
> 
> http://repos.rcn-ee.net/debian/pool/main/r/
> 
> Just run:
> 
> ```
> cd /opt/scripts/tools/
> git pull
> sudo ./update_kernel.sh
> sudo reboot
> sudo apt install rtl8821cu-modules-`uname -r`
> sudo reboot
> ```
> Regards,

@RobertCNelson 
Hooray! Hero!
It works!
Now Im going to check its function....
Thank you very much

closed 2019-02-02 by godspy16888