mkolender/rtl8821CU #27: Debian 4.9.144-3 (2019-02-02) build fail

Bsplesk, 2019-03-30:
Linux 4.9.0-8-amd64 #1 SMP Debian 4.9.144-3 (2019-02-02) x86_64 GNU/Linux

`/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/4.9.0-8-amd64/build M=/root/rtl8821CU  modules
make[1]: Entering directory '/usr/src/linux-headers-4.9.0-8-amd64'
/bin/sh: 1: bc: not found
  CC [M]  /root/rtl8821CU/core/rtw_cmd.o
  CC [M]  /root/rtl8821CU/core/rtw_security.o
  CC [M]  /root/rtl8821CU/core/rtw_debug.o
/root/rtl8821CU/core/rtw_debug.c: In function ‘dump_drv_version’:
/root/rtl8821CU/core/rtw_debug.c:50:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/root/rtl8821CU/core/rtw_debug.c:50:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
cc1: some warnings being treated as errors
/usr/src/linux-headers-4.9.0-8-common/scripts/Makefile.build:309: recipe for target '/root/rtl8821CU/core/rtw_debug.o' failed
make[4]: *** [/root/rtl8821CU/core/rtw_debug.o] Error 1
/usr/src/linux-headers-4.9.0-8-common/Makefile:1546: recipe for target '_module_/root/rtl8821CU' failed
make[3]: *** [_module_/root/rtl8821CU] Error 2
Makefile:152: recipe for target 'sub-make' failed
make[2]: *** [sub-make] Error 2
Makefile:8: recipe for target 'all' failed
make[1]: *** [all] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.9.0-8-amd64'
Makefile:1923: recipe for target 'modules' failed
make: *** [modules] Error 2
`

RobertCNelson, 2019-03-30:
@Bsplesk 
```
/bin/sh: 1: bc: not found
```

Install bc...

Regards,

Bsplesk, 2019-03-31:
close
sudo apt-get install build-essential -y
sudo apt-get install bc -y

closed 2019-03-31 by Bsplesk