mkolender/rtl8821CU #29: Compile error on Raspbian

buzzy, 2019-04-21:
I have changed the Makefile to build for RPI. Still I get error that "armv71" dir does not exist.
Adding "ARCH=arm" in front of the build command fixed this error, but introduces another compile error.

DKMS make.log for rtl8821CU-5.2.5.3 for kernel 4.14.98-v7+ (armv7l)
Sun 21 Apr 19:57:20 BST 2019
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.14.98-v7+/build M=/var/lib/dkms/rtl8821CU/5.2.5.3/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.98-v7+'
Makefile:646: arch/armv7l/Makefile: No such file or directory
make[1]: *** No rule to make target 'arch/armv7l/Makefile'.  Stop.
make[1]: Leaving directory '/usr/src/linux-headers-4.14.98-v7+'
Makefile:1923: recipe for target 'modules' failed
make: *** [modules] Error 2

DKMS make.log for rtl8821CU-5.2.5.3 for kernel 4.14.98-v7+ (armv7l)
Sun 21 Apr 19:59:56 BST 2019
make ARCH=arm CROSS_COMPILE= -C /lib/modules/4.14.98-v7+/build M=/var/lib/dkms/rtl8821CU/5.2.5.3/build  modules
make[1]: Entering directory '/usr/src/linux-headers-4.14.98-v7+'
  CC [M]  /var/lib/dkms/rtl8821CU/5.2.5.3/build/core/rtw_cmd.o
gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together
scripts/Makefile.build:326: recipe for target '/var/lib/dkms/rtl8821CU/5.2.5.3/build/core/rtw_cmd.o' failed
make[2]: *** [/var/lib/dkms/rtl8821CU/5.2.5.3/build/core/rtw_cmd.o] Error 1
Makefile:1535: recipe for target '_module_/var/lib/dkms/rtl8821CU/5.2.5.3/build' failed
make[1]: *** [_module_/var/lib/dkms/rtl8821CU/5.2.5.3/build] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-4.14.98-v7+'
Makefile:1923: recipe for target 'modules' failed
make: *** [modules] Error 2

buzzy, 2019-04-21:
I tried commenting out the "mfloat-abi=hard" in the Makefile. Ran "make clean" and tried compiling again. Same error.

buzzy, 2019-04-21:
root@raspberrypi:/opt/rtl8821CU# uname -a
Linux raspberrypi 4.14.98-v7+ #1200 SMP Tue Feb 12 20:27:48 GMT 2019 armv7l GNU/Linux

buzzy, 2019-04-22:
Got it working by doing a manual "make" and "make install". Seems the dkms-part is broken.

saulable, 2019-04-29:
Can you please explain how got it working with the manual install?

I'm getting gcc: error: unrecognized command line option '-mgeneral-regs-only'

buzzy, 2019-04-29:
I did not do anything special. Just ran "make" instead of compiling through dkms.

marki2239, 2019-06-04:
hi  im having the same problem  and manual sudo make doesnt work..  Any suggestions?

sunzone93, 2019-08-23:
I get the same error
> gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together

My environment: Orange Pi Zero, Kernal 4.19.62-sunxi, Ubuntu 18.04.3 boinic

> I tried commenting out the "mfloat-abi=hard" in the Makefile. Ran "make clean" and tried compiling again. Same error.

I find the same error too.

I run just "make" too. But the problem persists.

sunzone93, 2019-08-23:
I managed to solve the compilation issue.
Please refer the following fork I created. You can use the below forked repository to compile without "gcc: error: -mfloat-abi=soft and -mfloat-abi=hard may not be used together" error.

https://github.com/sunzone93/rtl8821CU

All the best!

shivan, 2019-11-12:
@sunzone93 was this on rpi3b?
I'm trying your version on rpi3b... but still some errors.
First in line 1924 there were spaces instead of tabs

sunzone93, 2020-03-19:
@shivan thanks for the check. I fixed the spaces.
I tested this on Orange Pi Zero. Haven't tested on rpi3b yet.
I will update you soon after testing on rpi3b

shivan, 2020-03-19:
@sunzone93 would be great to have this working, as I already tried several other sources, with bad luck :-(

sunzone93, 2020-03-23:
Hi @shivan I managed to compile the library for RPi3b with following steps. 
1) install kernel headers
```
sudo apt install raspberrypi-kernel-headers
```
2) Use following commands in source directory
```
make
sudo make install
sudo modprobe 8821cu
```
*Don't update the Makefile for CONFIG_PLATFORM_ARM_RPI3 = y. Keep the default options. When this option is enabled (error: unrecognized command line option ‘-mgeneral-regs-only’) gcc error occurs. You may follow [this](https://github.com/raspberrypi/linux/issues/2682).

3) Check whether the driver is installed properly
```
ls /lib/modules/$(uname -r)/kernel/drivers/net/wireless
```
Make sure `8821cu.ko` file present on that directory

I updated my repository to fix several bugs. So you may re-clone my repo again!

Please let me know how it goes~

shivan, 2020-03-26:
Hi,

thanks for your update. Now I get this:

```
pi@rover:~/white $ sudo apt install raspberrypi-kernel-headers
Reading package lists... Done
Building dependency tree
Reading state information... Done
raspberrypi-kernel-headers is already the newest version (1.20200212-1).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
pi@rover:~/white $ git clone https://github.com/whitebatman2/rtl8821CU.git
Cloning into 'rtl8821CU'...
remote: Enumerating objects: 1, done.
remote: Counting objects: 100% (1/1), done.
remote: Total 666 (delta 0), reused 0 (delta 0), pack-reused 665
Receiving objects: 100% (666/666), 3.31 MiB | 2.10 MiB/s, done.
Resolving deltas: 100% (246/246), done.
pi@rover:~/white $ cd rtl8821CU/
pi@rover:~/white/rtl8821CU $ ls
clean  core  dkms.conf  hal  ifcfg-wlan0  include  Kconfig  LICENSE  Makefile  os_dep  platform  README.md  rtl8821c.mk  runwpa  wlan0dhcp
pi@rover:~/white/rtl8821CU $ make
make ARCH=armv7l CROSS_COMPILE= -C /lib/modules/4.19.93-v7+/build M=/home/pi/white/rtl8821CU  modules
make[1]: *** /lib/modules/4.19.93-v7+/build: No such file or directory.  Stop.
make: *** [Makefile:1923: modules] Error 2
pi@rover:~/white/rtl8821CU $
```

What's the problem now? looks like it is missing some libraries?

sunzone93, 2020-03-26:
@shivan please clone [my fork](https://github.com/sunzone93/rtl8821CU) and test.
```
git clone https://github.com/sunzone93/rtl8821CU.git
```

shivan, 2020-03-26:
@sunzone93 oops, sorry.

Ok, used your fork.

```
pi@rover:~/white $ git clone https://github.com/sunzone93/rtl8821CU.git
Cloning into 'rtl8821CU'...
remote: Enumerating objects: 36, done.
remote: Counting objects: 100% (36/36), done.
remote: Compressing objects: 100% (34/34), done.
remote: Total 687 (delta 19), reused 3 (delta 2), pack-reused 651
Receiving objects: 100% (687/687), 3.35 MiB | 2.46 MiB/s, done.
Resolving deltas: 100% (257/257), done.
pi@rover:~/white $ cd rtl8821CU/
pi@rover:~/white/rtl8821CU $ ls
clean  core  dkms.conf  hal  ifcfg-wlan0  include  Kconfig  LICENSE  Makefile  os_dep  platform  README.md  rtl8821c.mk  runwpa  wlan0dhcp
pi@rover:~/white/rtl8821CU $ make
sudo cp /lib/modules/4.19.93-v7+/build/arch/arm/Makefile /lib/modules/4.19.93-v7+/build/arch/arm/Makefile.202003260645
cp: cannot stat '/lib/modules/4.19.93-v7+/build/arch/arm/Makefile': No such file or directory
make: *** [Makefile:1923: modules] Error 1
pi@rover:~/white/rtl8821CU $
```

But not better. Why is it missing that directory. Solution in the web is mostly to install raspberrypi-kernel-headers. Which is already installed.

sunzone93, 2020-03-26:
@shivan may be follow [this](https://raspberrypi.stackexchange.com/questions/50240/missing-build-file-when-building-for-rtl8812au-driver) and see? I will take a look into it over this weekend.

sunzone93, 2020-03-26:
can you try to compile as the superuser (after ```sudo su```)? or try with ```sudo make``` ?
I am not in a proper environment to test rn.

sunzone93, 2020-03-26:
if you can, also test after changing [Makefile line 1923](https://github.com/sunzone93/rtl8821CU/blob/95a54a79bfcf9fb571ea58ab2121a9d6234aec57/Makefile#L1923) with ```sudo cp '/lib/modules/$(KVER)/build/arch/arm/Makefile' /lib/modules/$(KVER)/build/arch/arm/Makefile.$(shell date +%Y%m%d%H%M)```
ref: [1](https://stackoverflow.com/questions/34254164/what-is-cp-cannot-stat-error-in-unix-i-get-this-error-when-trying-to-copy-thin), [2](https://askubuntu.com/questions/1079647/cp-cannot-stat-no-such-file-or-directory-exists)

shivan, 2020-03-26:
mhm something interesting...

`/lib/modules/4.19.97-v7+`

exists.

but uname -r returns
`4.19.93-v7+`

so my modules are newer than the running kernel?

I'll reboot...

shivan, 2020-03-26:
wohoo reboot... now "make" is doing something :)

my lession for today:
always do a reboot after "apt-get update;apt-get upgrade;apt-get dist-upgrade" ...

ZhipengRobot, 2020-06-22:
> Hi @shivan I managed to compile the library for RPi3b with following steps.
> 
> 1. install kernel headers
> 
> ```
> sudo apt install raspberrypi-kernel-headers
> ```
> 
> 1. Fork the git library
> 
> ```
> git clone https://github.com/sunzone93/rtl8821CU.git
> ```
> 
> 1. Use following commands in source the directory (cd rtl8821CU)
> 
> ```
> make
> sudo make install
> sudo modprobe 8821cu
> ```
> 
> *Don't update the Makefile for CONFIG_PLATFORM_ARM_RPI3 = y. Keep the default options. When this option is enabled (error: unrecognized command line option ‘-mgeneral-regs-only’) gcc error occurs. You may follow [this](https://github.com/raspberrypi/linux/issues/2682).
> 
> 1. Check whether the driver is installed properly
> 
> ```
> ls /lib/modules/$(uname -r)/kernel/drivers/net/wireless
> ```
> 
> Make sure `8821cu.ko` file present on that directory
> 
> I updated my repository to fix several bugs. So you may re-clone my repo again!
> 
> Please let me know how it goes~

I tried sunzone93's solution on raspi CM3, it works!
Thank you very much!

AverageMaker, 2020-08-14:
I seem to get near the end of sunzone's method and I get an error saying that it can't locate the 8821cu.ko file. Anyone else had this or know the fix?

sunzone93, 2020-08-14:
> I seem to get near the end of sunzone's method and I get an error saying that it can't locate the 8821cu.ko file. Anyone else had this or know the fix?

@AverageMaker please share your test environment

AverageMaker, 2020-08-14:
> @AverageMaker please share your test environment
I'm using a Raspberry Pi 3B+ with a clean (new) copy of the latest Raspberry Pi OS installed from a new NOOBS image (3.4.0). I'll need to check kernel versions when I get home. Nothing else is attached apart from keyboard and mouse. Using this RTL8811cu adapter: https://thepihut.com/products/raspberry-pi-dual-band-5ghz-2-4ghz-usb-wifi-adapter-with-antenna

AverageMaker, 2020-09-02:
> > I seem to get near the end of sunzone's method and I get an error saying that it can't locate the 8821cu.ko file. Anyone else had this or know the fix?
> 
> @AverageMaker please share your test environment

Also noticed that this is likely to do with the latest (20th Aug) Raspberry Pi OS version shipping with Linux kernel 5.4.51, whereas previously this was on 4.19.118. Since that kernel change the driver no longer works.,

I'm no expert on drivers and kernels - is this something that can easily be resolved? Do we need to wait for Realtek to release an updated driver or can we tweak the current driver to work with the newer Linux kernel?

sunzone93, 2020-11-26:
Yes. Currently the driver doesn't support the new kernel. 
I will try to port it in my free time.

sp1d3rino, 2021-04-24:
> I have changed the Makefile to build for RPI. Still I get error that "armv71" dir does not exist.
> Adding "ARCH=arm" in front of the build command fixed this error, but introduces another compile [...]

1. vi Makefile
2. search for -mfloat-abi field and change hard to soft
3. save & restart make