mkolender/rtl8821CU #30: Compiling error on Raspbian kernel version  4.14.98 (Raspberry 3 Model B+)

AngelsWillRule, 2019-05-12:
Well, compiling this module was giving me an error till i added in the Makefile an extra flag ( EXTRA_CFLAGS += -Wno-error=date-time ). Compiling now is ok.

marki2239, 2019-06-04:
hi  im having problems compiling under 4.14.98   I'm getting the following error

Pls help!

pi@raspberrypi:~/rtl8821CU $ sudo make
make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.98-v7+/build M=/home/pi/rtl8821CU  modules
make[1]: Entering directory '/lib/modules/4.14.98-v7+/build'
make[1]: *** No rule to make target 'modules'.  Stop.
make[1]: Leaving directory '/lib/modules/4.14.98-v7+/build'
Makefile:1923: recipe for target 'modules' failed
make: *** [modules] Error 2

AngelsWillRule, 2019-06-04:
> hi im having problems compiling under 4.14.98 I'm getting the following error
>
> Pls help!
>
> pi@raspberrypi:~/rtl8821CU $ sudo make
> make ARCH=arm64 CROSS_COMPILE= -C /lib/modules/4.14.98-v7+/build [...]

Hello Marki. All i did is this.

```
sudo apt update
sudo apt install build-essential git dkms
git clone https://github.com/whitebatman2/rtl8821CU.git
DRV_NAME=rtl8821CU
DRV_VERSION=5.2.5.3
sudo mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
cd $DRV_NAME

```
Now that you are in the driver directory, edit the MakeFile and add this CFLAG to the others present at the top of the MakeFile:

`EXTRA_CFLAGS += -Wno-error=date-time`

And change these MakeFile entries to ( look for them cause they are already present in the MakeFile):
```
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM_RPI3 = y

```
and then:

```
git archive master | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
sudo dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
sudo dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
```

I hope it helps!

PepperNickel, 2019-06-05:
Thank you for the wealth of info. I'm a beginner and stuck on trying to get this drive up and going on my pi 3 b+. I've made it this far but I'm stuck on trying to edit the MakeFile. I've found the MakeFile in the rtl8821CU folder but I don't know the best way to edit it. To repay for the help I'll write up an Install to Raspberry Pi 3 for Beginners somewhere on here so to limit further questions for a while.

AngelsWillRule, 2019-06-06:
> Thank you for the wealth of info. I'm a beginner and stuck on trying to get this drive up and going on my pi 3 b+. I've made it this far but I'm stuck on trying to edit the MakeFile. I've found the [...]

Hello PepperNickel. 
The best and easiest way to edit a file using the terminal is nano. If it's not installed by default, digit:
`sudo apt-get install -y nano`
Once you are in the driver directory, digit
`sudo nano MakeFile`
To move through the file, use the arrows. To save the file, use CTRL+O and confirm pressing ENTER. To exit the edit of the file, use CTRL+X.
Have a nice day.

marki2239, 2019-06-06:
Hi!

 

I followed the procedures and still had compilation issued.  After doing some reading I saw that you could compile for RPI even if I was using  RPI3+  and it worked.   Wouldn’t go into monintor mode.   Any suggestions on how to enable monitor mode?

 

But if u run into compilation issued try RPI and not RPI3

 

Thanks

 

Mark

PepperNickel, 2019-06-06:
So far I've edited MakeFile and ran everything. The Errors shown 
`Your kernel headers for kernel 4.19.42-v7+ cannot be found at
/lib/modules/4.19.42-v7+/build or /lib/modules/4.19.42-v7+/source.
`.

I've never done anything with Linux in my life. I think I'll just watch some basic instruction videos and come back and give it a go later. After I learn a few more basic operations.

Thank you

AngelsWillRule, 2019-06-07:
> So far I've edited MakeFile and ran everything. The Errors shown
> `Your kernel headers for kernel 4.19.42-v7+ cannot be found at /lib/modules/4.19.42-v7+/build or /lib/modules/4.19.42-v7+/source. `. [...]

Try with

`sudo apt-get install raspberrypi-kernel-headers`

Cheers

PepperNickel, 2019-08-12:
I've been trying for a quite some time. Maybe someone can see what I'm doing wrong. So here's my path.

1.  
```
sudo apt update
sudo apt install build-essential git dkms
git clone https://github.com/whitebatman2/rtl8821CU.git
```
2. 
`
sudo nano MakeFile
`
Add:
`
EXTRA_CFLAGS += -Wno-error=date-time
`
Edit:
`
CONFIG_PLATFORM_I386_PC = n
CONFIG_PLATFORM_ARM_RPI = n
CONFIG_PLATFORM_ARM_RPI3 = y
`
3. Closed and reopened command promt
`
sudo apt-get install raspberrypi-kernel-headers
`

And before compiling again.
`
sudo make clean
`

4.
````
DRV_NAME=rtl8821CU
DRV_VERSION=5.2.5.3
sudo mkdir /usr/src/${DRV_NAME}-${DRV_VERSION}
cd $DRV_NAME
````
then
````
git archive master | sudo tar -x -C /usr/src/${DRV_NAME}-${DRV_VERSION}
sudo dkms add -m ${DRV_NAME} -v ${DRV_VERSION}
sudo dkms build -m ${DRV_NAME} -v ${DRV_VERSION}
sudo dkms install -m ${DRV_NAME} -v ${DRV_VERSION}
````
________________________________________________________________
Still getting the "Your kernel headers for kernel blablabla cannot be found". I know I can find another use for this wifi dongle if this doesn't work out. Thank you for all the help so far!

marki2239, 2019-08-13:
Try compiling as RPI not RPI3

 

Also you need to do a git commit if youre going to use dkms otherwise it wont keep your changed to Makefile.

 

Worked for me.

 

Mark

AngelsWillRule, 2019-08-13:
Try to use the command 
`apt-get install --reinstall raspberrypi-bootloader  git dkms build-essential raspberrypi-kernel raspberrypi-kernel-headers`
and then repeat the procedure.
Otherwise, i found another working driver for this chipset.
Anyway, give the command provided before proceding cause you have to solve first that problem.
The alternative driver is found at the page https://github.com/brektrou/rtl8821CU .
Follow the instructions and before compiling give this commands:
`sudo cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)`
`sudo sed -i 's/-msoft-float//' /lib/modules/$(uname -r)/build/arch/arm/Makefile`
`sudo ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l`
The compiling takes about 10 minutes on the Raspberry. After that, try to plug your usb wifi dongle. If it does not work, reboot the Raspberry. Hope it helps.

PepperNickel, 2019-08-13:
I believe I've been editing the wrong MakeFile... I see an rtl8821CU folder under /home/pi and another rtl8821CU in /usr/src/rtl8821CU-5.2.5.3 I can edit the MakeFile /home/pi but not the one down in src.

So my problem is my limited knowledge as to open and edit the protected MakeFile. I did see that my edits to the /home/pi MakeFile did save. So some progress.

Also should I uninstall this driver before trying the other one?

I gave the new driver a try. Got as far as...
`` 
sudo cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)
``
And received
``
cp: cannot stat '/lib/modules/4.19.57-v7+/build/arch/arm/Makefile': No such file or directory
``

I may be too much  of a noob for this but I'm not ready to give up. I have also gone as far as reinstalling rasbian. Thinking I may have made files I can't undo.
Thank you!

AngelsWillRule, 2019-08-13:
yes, enter the directory and uninstall the previous driver using the make clean command.
Be sure to have installed all the packets needed for the compilation with the command i gave you with 

`sudo apt-get install --reinstall raspberrypi-bootloader git dkms build-essential raspberrypi-kernel raspberrypi-kernel-headers`

I'm really optimistic that with the commands i gave you to install the new driver you will your wifi dongle work. 

Bye.

AngelsWillRule, 2019-08-13:
Anyway, i'm using 4.14.98-v7+ kernel version. So, maybe this is the problem.

AngelsWillRule, 2019-08-18:
> I believe I've been editing the wrong MakeFile... I see an rtl8821CU folder under /home/pi and another rtl8821CU in /usr/src/rtl8821CU-5.2.5.3 I can edit the MakeFile /home/pi but not the one down in src.

So my problem is my limited knowledge as to open and edit the protected MakeFile. I did see that my edits to the /home/pi MakeFile did save. So some progress.

Also should I uninstall this driver before trying the other one?

I gave the new driver a try. Got as far as...
sudo cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)
And received
cp: cannot stat '/lib/modules/4.19.57-v7+/build/arch/arm/Makefile': No such file or directory

I may be too much of a noob for this but I'm not ready to give up. I have also gone as far as reinstalling rasbian. Thinking I may have made files I can't undo.
Thank you!

Hello PepperNickel. I'm now on kernel 4.19.58-v7 and the compiling works flawlessly.
I suggest to completely reinstall your Raspbian using the latest version from the following link:
https://www.raspberrypi.org/downloads/raspbian/

Once installed, give the following commands:
`sudo su`

`apt-get update`

`apt-get install  git dkms build-essential  raspberrypi-kernel-headers bc`

`mkdir -p ~/build`

`cd ~/build`

`git clone https://github.com/brektrou/rtl8821CU.git`

`cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)`

`sed -i 's/-msoft-float//' /lib/modules/$(uname -r)/build/arch/arm/Makefile`

`ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l`

`./dkms-install.sh`

You can't be wrong! I have just tested everything.

Cheers.

marki2239, 2019-08-19:
58 worked for me too I could not get it to work on 98

And make sure you do a hit commit after you make the changes otherwise the sims will not work

Mark

QiuYukang, 2020-06-29:
> Hello PepperNickel. I'm now on kernel 4.19.58-v7 and the compiling works flawlessly.
> I suggest to completely reinstall your Raspbian using the latest version from the following link:
> https://www.raspberrypi.org/downloads/raspbian/
> 
> Once installed, give the following commands:
> `sudo su`
> 
> `apt-get update`
> 
> `apt-get install git dkms build-essential raspberrypi-kernel-headers bc`
> 
> `mkdir -p ~/build`
> 
> `cd ~/build`
> 
> `git clone https://github.com/brektrou/rtl8821CU.git`
> 
> `cd rtl8821CU`
> 
> `cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)`
> 
> `sed -i 's/-msoft-float//' /lib/modules/$(uname -r)/build/arch/arm/Makefile`
> 
> `ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l`
> 
> `./dkms-install.sh`
> 
> You can't be wrong! I have just tested everything.
> 
> Cheers.

It also works on Raspberry 4 Model B+, thank you very much!

jwest75674, 2020-07-08:
Wanted to add to this old thread. It took a few days, but I got these cheapo adapters working and wanted to share what I had learned for anyone else that might need it.

Raspberry Pi 4b

the adapter up, though it was labelled as a different chipset, confirmed directly to be rtl8811**CU**
--> It's also mislabelled from the manufacturer as a different chipset. Thanks China.

I followed @AngelsWillRule 's solution above, but it still fails at the ./dkms-install.sh with
`gcc: error: unrecognized command line option ‘-mgeneral-regs-only’`

I recognized this error from the airmon-ng/rtl8812AU git, where they added the `ARCH=arm` to dkms.conf and dkms-install.sh's commands, but this didn't seem to help.

I have a few of these adapters, and wouldn't you know, more trouble as a result.
These adapters seem to have a couple of different IDs, I've seen:
0bda:5411
0bda:0411 
0bda:c811

As there are a few different IDs as above, I added all to `os_dep/linux/usb_intf.c`, but duplicating a line under the RTL8821CU header, and changing out the id on the new duplicate lines (being careful to match the 0xc811 format)

I was still having troubles and found myself working through the solution seen above by @PepperNickel :
https://github.com/whitebatman2/rtl8821CU/issues/30#issuecomment-520615423

Long story short, these two solutions have what is needed to get this fixed. +1 from me, testing further now!

jwest75674, 2020-07-08:
Spoke too soon. When connecting these up (I have a dozen of them), I am getting some issues....

```
[  179.312527] usb 1-1.2.4: new high-speed USB device number 71 using xhci_hcd
[  179.323209] ------------[ cut here ]------------
[  179.325013] WARNING: CPU: 0 PID: 78 at /home/pi/rtl/rtl8821CU/os_dep/linux/os_intfs.c:3105 rtw_os_ndevs_register+0xfc/0x144 [8821cu]
[  179.327670] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp appletalk psnap llc ax25 pl2303 usbserial bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) rpivid_mem 8821cu(O) vc4 brcmfmac drm_kms_helper v3d gpu_sched snd_soc_core snd_bcm2835(C) brcmutil snd_compress drm snd_pcm_dmaengine snd_pcm drm_panel_orientation_quirks raspberrypi_hwmon hwmon snd_timer snd syscopyarea sysfillrect sysimgblt cfg80211 fb_sys_fops rfkill uio_pdrv_genirq uio ip_tables x_tables ipv6
[  179.340789] CPU: 0 PID: 78 Comm: kworker/0:2 Tainted: G      D WC O      4.19.118-v7l+ #1311
[  179.343131] Hardware name: BCM2835
[  179.345410] Workqueue: usb_hub_wq hub_event
[  179.347693] [<c0212e0c>] (unwind_backtrace) from [<c020d5e8>] (show_stack+0x20/0x24)
[  179.350081] [<c020d5e8>] (show_stack) from [<c09c2568>] (dump_stack+0xe0/0x124)
[  179.352508] [<c09c2568>] (dump_stack) from [<c0222540>] (__warn+0x104/0x11c)
[  179.354954] [<c0222540>] (__warn) from [<c0222690>] (warn_slowpath_null+0x50/0x58)
[  179.357894] [<c0222690>] (warn_slowpath_null) from [<bf955594>] (rtw_os_ndevs_register+0xfc/0x144 [8821cu])
[  179.363841] [<bf955594>] (rtw_os_ndevs_register [8821cu]) from [<bf955650>] (rtw_os_ndevs_init+0x34/0x4c [8821cu])
[  179.370219] [<bf955650>] (rtw_os_ndevs_init [8821cu]) from [<bf9575f0>] (rtw_drv_init+0x4a4/0x674 [8821cu])
[  179.376277] [<bf9575f0>] (rtw_drv_init [8821cu]) from [<c07a2650>] (usb_probe_interface+0x110/0x294)
[  179.382071] [<c07a2650>] (usb_probe_interface) from [<c07184e4>] (really_probe+0x228/0x2d8)
[  179.385138] [<c07184e4>] (really_probe) from [<c0718768>] (driver_probe_device+0x70/0x1ac)
[  179.388229] [<c0718768>] (driver_probe_device) from [<c0718a78>] (__device_attach_driver+0x98/0xa8)
[  179.394378] [<c0718a78>] (__device_attach_driver) from [<c0716328>] (bus_for_each_drv+0x88/0xcc)
[  179.397577] [<c0716328>] (bus_for_each_drv) from [<c0718238>] (__device_attach+0xe0/0x144)
[  179.400753] [<c0718238>] (__device_attach) from [<c0718aa4>] (device_initial_probe+0x1c/0x20)
[  179.403907] [<c0718aa4>] (device_initial_probe) from [<c07174fc>] (bus_probe_device+0x94/0x9c)
[  179.407037] [<c07174fc>] (bus_probe_device) from [<c0714f88>] (device_add+0x468/0x600)
[  179.410131] [<c0714f88>] (device_add) from [<c07a0220>] (usb_set_configuration+0x51c/0xa5c)
[  179.413203] [<c07a0220>] (usb_set_configuration) from [<c07abc9c>] (generic_probe+0x3c/0x88)
[  179.416250] [<c07abc9c>] (generic_probe) from [<c07a2528>] (usb_probe_device+0x34/0x4c)
[  179.419260] [<c07a2528>] (usb_probe_device) from [<c07184e4>] (really_probe+0x228/0x2d8)
[  179.422234] [<c07184e4>] (really_probe) from [<c0718768>] (driver_probe_device+0x70/0x1ac)
[  179.425184] [<c0718768>] (driver_probe_device) from [<c0718a78>] (__device_attach_driver+0x98/0xa8)
[  179.430958] [<c0718a78>] (__device_attach_driver) from [<c0716328>] (bus_for_each_drv+0x88/0xcc)
[  179.434005] [<c0716328>] (bus_for_each_drv) from [<c0718238>] (__device_attach+0xe0/0x144)
[  179.437045] [<c0718238>] (__device_attach) from [<c0718aa4>] (device_initial_probe+0x1c/0x20)
[  179.440095] [<c0718aa4>] (device_initial_probe) from [<c07174fc>] (bus_probe_device+0x94/0x9c)
[  179.443182] [<c07174fc>] (bus_probe_device) from [<c0714f88>] (device_add+0x468/0x600)
[  179.446276] [<c0714f88>] (device_add) from [<c07964d4>] (usb_new_device+0x27c/0x450)
[  179.449359] [<c07964d4>] (usb_new_device) from [<c0797fe8>] (hub_event+0xdc0/0x11d8)
[  179.452407] [<c0797fe8>] (hub_event) from [<c023e080>] (process_one_work+0x170/0x458)
[  179.455424] [<c023e080>] (process_one_work) from [<c023e3c4>] (worker_thread+0x5c/0x5a4)
[  179.458407] [<c023e3c4>] (worker_thread) from [<c02446f8>] (kthread+0x138/0x168)
[  179.461346] [<c02446f8>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[  179.464235] Exception stack(0xef2d7fb0 to 0xef2d7ff8)
[  179.467058] 7fa0:                                     00000000 00000000 00000000 00000000
[  179.469907] 7fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  179.472693] 7fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  179.476159] ---[ end trace 9c150faee6d5db55 ]---
[  179.480771] RTW: WARN free_recv_skb_queue not empty, 8
[  179.512946] ------------[ cut here ]------------
[  179.515686] WARNING: CPU: 2 PID: 1243 at net/wireless/sme.c:756 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
[  179.520843] Modules linked in: cmac bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp appletalk psnap llc ax25 pl2303 usbserial bcm2835_codec(C) v4l2_mem2mem videobuf2_dma_contig bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) rpivid_mem 8821cu(O) vc4 brcmfmac drm_kms_helper v3d gpu_sched snd_soc_core snd_bcm2835(C) brcmutil snd_compress drm snd_pcm_dmaengine snd_pcm drm_panel_orientation_quirks raspberrypi_hwmon hwmon snd_timer snd syscopyarea sysfillr [... cut ...]
[  237.504975] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  237.944939] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  237.946124] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  238.014711] RTW: ERROR _halmac_reg_write_32: I/O FAIL!
[  238.015775] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  238.016752] RTW: ERROR _halmac_reg_write_8: I/O FAIL!
[  238.017649] RTW: ERROR _halmac_reg_write_16: I/O FAIL!
[  238.018583] RTW: rtw_cmd_thread: DriverStopped(False) SurpriseRemoved(True) break at line 580

```

jwest75674, 2020-10-11:
Correction, had nothing to do with power issues.

Instead, after weeks of periodic searching, I finally stumbled across these resources:
https://github.com/lwfinger/rtl8723bu/issues/131
https://forum.armbian.com/topic/8241-wifi-access-point-hostapd-with-kernel-418-and-419/

Specifically the portion about adding 
`sinfo.pertid = NULL;`

above
```
sinfo.pertid = 0;  // Somewhere here (before cfg80211_new_sta(…)) insert this line.
sinfo.filled = 0;
sinfo.assoc_req_ies = pmgmt_frame + WLAN_HDR_A3_LEN + ie_offset;
sinfo.assoc_req_ies_len = frame_len - WLAN_HDR_A3_LEN - ie_offset;
cfg80211_new_sta(ndev, GetAddr2Ptr(pmgmt_frame), &sinfo, GFP_ATOMIC);
```
within the file `./os_dep/linux/ioctl_cfg80211.c`

appears that it may have resolved the above noted issue.