mkolender/rtl8821CU #35: Driver crash after a few hours of uptime

robeving, 2019-09-04:
`[293120.364476] ------------[ cut here ]------------
[293120.364774] WARNING: CPU: 2 PID: 387 at /var/lib/dkms/rtl8821CU/5.2.5.3/build/hal/rtl8821c/rtl8821c_phy.c:864 rtl8821c_switch_chnl_and_set_bw+0x150/0x204 [8821cu]
[293120.364783] Modules linked in: nft_counter nft_chain_nat_ipv4 ipt_MASQUERADE nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink tun rfcomm bnep hci_uart btbcm serdev bluetooth ecdh_generic appletalk psnap ax25 sg evdev r820t rtl2832 i2c_mux dvb_usb_rtl28xxu dvb_usb_v2 dvb_core uas 8821cu(O) raspberrypi_hwmon hwmon brcmfmac snd_bcm2835(C) brcmutil bcm2835_codec(C) bcm2835_v4l2(C) snd_pcm v4l2_mem2mem bcm2835_mmal_vchiq(C) snd_timer videobuf2_dma_contig v4l2_common videobuf2_vmalloc snd videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed 8021q garp stp llc 8192cu sha256_generic cfg80211 rfkill ip_tables x_tables ipv6
[293120.364957] CPU: 2 PID: 387 Comm: RTW_CMD_THREAD Tainted: G        WC O      4.19.66-v7+ #1253
[293120.364961] Hardware name: BCM2835
[293120.364979] [<80111f38>] (unwind_backtrace) from [<8010d4b0>] (show_stack+0x20/0x24)
[293120.364988] [<8010d4b0>] (show_stack) from [<808191e0>] (dump_stack+0xd4/0x118)
[293120.364997] [<808191e0>] (dump_stack) from [<801209c8>] (__warn+0x104/0x11c)
[293120.365007] [<801209c8>] (__warn) from [<80120b18>] (warn_slowpath_null+0x50/0x58)
[293120.365267] [<80120b18>] (warn_slowpath_null) from [<7f66a6c0>] (rtl8821c_switch_chnl_and_set_bw+0x150/0x204 [8821cu])
[293120.365707] [<7f66a6c0>] (rtl8821c_switch_chnl_and_set_bw [8821cu]) from [<7f66a930>] (rtl8821c_handle_sw_chnl_and_set_bw+0x1bc/0x1c0 [8821cu])
[293120.366136] [<7f66a930>] (rtl8821c_handle_sw_chnl_and_set_bw [8821cu]) from [<7f66a970>] (rtl8821c_set_channel_bw+0x3c/0x44 [8821cu])
[293120.366563] [<7f66a970>] (rtl8821c_set_channel_bw [8821cu]) from [<7f6416e8>] (rtw_hal_set_chnl_bw+0x130/0x150 [8821cu])
[293120.366981] [<7f6416e8>] (rtw_hal_set_chnl_bw [8821cu]) from [<7f5fbdd0>] (set_channel_bwmode+0x90/0xcc [8821cu])
[293120.367394] [<7f5fbdd0>] (set_channel_bwmode [8821cu]) from [<7f5f6ddc>] (site_survey+0x34/0x1d8 [8821cu])
[293120.367806] [<7f5f6ddc>] (site_survey [8821cu]) from [<7f5f798c>] (sitesurvey_cmd_hdl+0x774/0x7b4 [8821cu])
[293120.368207] [<7f5f798c>] (sitesurvey_cmd_hdl [8821cu]) from [<7f5ce074>] (rtw_cmd_thread+0x218/0x3b4 [8821cu])
[293120.368415] [<7f5ce074>] (rtw_cmd_thread [8821cu]) from [<8014253c>] (kthread+0x138/0x168)
[293120.368427] [<8014253c>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[293120.368430] Exception stack(0xb4549fb0 to 0xb4549ff8)
[293120.368435] 9fa0:                                     00000000 00000000 00000000 00000000
[293120.368440] 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[293120.368444] 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[293120.368451] ---[ end trace bbeb82e268bc4182 ]---`

robeving, 2019-09-04:
>addr2line -e /lib/modules/4.19.66-v7+/updates/dkms/8821cu.ko 7f66a6c0
??:0

I'll see if I can compile an unstripped version of this and get a line number

Denny-89, 2019-10-16:
I had similar (if not same) problems on a Pi Zero /w RTL8811CU after Raspbian devs moved the kernel from 4.14 to 4.19. The only thing that fixed it was getting the newer driver version from https://github.com/whitebatman2/rtl8821CU/issues/23.