mkolender/rtl8821CU #37: Compile failed on kali 5.2.0-kali3-amd64

Archerkong, 2020-01-02:
make
/bin/sh: 1: bc: not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.2.0-kali3-amd64/build M=/opt/rtl8821CU  modules
make[1]: 进入目录“/usr/src/linux-headers-5.2.0-kali3-amd64”
/bin/sh: 1: bc: not found
  CC [M]  /opt/rtl8821CU/core/rtw_debug.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_debug.c:22:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_debug.c:22:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
/opt/rtl8821CU/core/rtw_debug.c: In function ‘dump_drv_version’:
/opt/rtl8821CU/core/rtw_debug.c:50:62: error: macro "__DATE__" might prevent reproducible builds [-Werror=date-time]
  RTW_PRINT_SEL(sel, "build time: %s %s\n", __DATE__, __TIME__);
                                                              ^
/opt/rtl8821CU/core/rtw_debug.c:50:62: error: macro "__TIME__" might prevent reproducible builds [-Werror=date-time]
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-5.2.0-kali3-common/scripts/Makefile.build:284：/opt/rtl8821CU/core/rtw_debug.o] 错误 1
make[3]: *** [/usr/src/linux-headers-5.2.0-kali3-common/Makefile:1610：_module_/opt/rtl8821CU] 错误 2
make[2]: *** [Makefile:179：sub-make] 错误 2
make[1]: *** [Makefile:8：all] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.2.0-kali3-amd64”
make: *** [Makefile:1923：modules] 错误 2

Archerkong, 2020-01-02:
that problem i had solve ,but i fine another problem 
make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/5.2.0-kali3-amd64/build M=/opt/rtl8821CU  modules
make[1]: 进入目录“/usr/src/linux-headers-5.2.0-kali3-amd64”
  CC [M]  /opt/rtl8821CU/core/rtw_cmd.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_cmd.c:22:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_cmd.c:22:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /opt/rtl8821CU/core/rtw_security.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_security.c:22:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_security.c:22:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /opt/rtl8821CU/core/rtw_debug.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_debug.c:22:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_debug.c:22:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /opt/rtl8821CU/core/rtw_io.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_io.c:52:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_io.c:52:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /opt/rtl8821CU/core/rtw_ioctl_query.o
In file included from /opt/rtl8821CU/include/drv_types.h:35,
                 from /opt/rtl8821CU/core/rtw_ioctl_query.c:22:
/opt/rtl8821CU/include/wifi.h:1019: warning: "IEEE80211_MAX_AMPDU_BUF" redefined
 #define IEEE80211_MAX_AMPDU_BUF 0x40
 
In file included from /opt/rtl8821CU/include/osdep_service_linux.h:86,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/core/rtw_ioctl_query.c:22:
/usr/src/linux-headers-5.2.0-kali3-common/include/linux/ieee80211.h:1441: note: this is the location of the previous definition
 #define IEEE80211_MAX_AMPDU_BUF  0x100
 
  CC [M]  /opt/rtl8821CU/core/rtw_ioctl_set.o
[... 362 lines cut ...]
/opt/rtl8821CU/os_dep/osdep_service.c:2171:11: error: incompatible type for argument 1 of ‘set_fs’
    set_fs(get_ds());
           ^~~~~~~~
In file included from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/uaccess.h:11,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/crypto.h:21,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/crypto/hash.h:11,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/uio.h:10,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/socket.h:8,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/compat.h:15,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/ethtool.h:17,
                 from /usr/src/linux-headers-5.2.0-kali3-common/include/linux/netdevice.h:37,
                 from /opt/rtl8821CU/include/osdep_service_linux.h:35,
                 from /opt/rtl8821CU/include/osdep_service.h:47,
                 from /opt/rtl8821CU/include/drv_types.h:32,
                 from /opt/rtl8821CU/os_dep/osdep_service.c:24:
/usr/src/linux-headers-5.2.0-kali3-common/arch/x86/include/asm/uaccess.h:29:40: note: expected ‘mm_segment_t’ {aka ‘struct <anonymous>’} but argument is of type ‘int’
 static inline void set_fs(mm_segment_t fs)
                           ~~~~~~~~~~~~~^~
cc1: some warnings being treated as errors
make[4]: *** [/usr/src/linux-headers-5.2.0-kali3-common/scripts/Makefile.build:284：/opt/rtl8821CU/os_dep/osdep_service.o] 错误 1
make[3]: *** [/usr/src/linux-headers-5.2.0-kali3-common/Makefile:1610：_module_/opt/rtl8821CU] 错误 2
make[2]: *** [Makefile:179：sub-make] 错误 2
make[1]: *** [Makefile:8：all] 错误 2
make[1]: 离开目录“/usr/src/linux-headers-5.2.0-kali3-amd64”
make: *** [Makefile:1923：modules] 错误 2

stockshuang, 2020-02-06:
Archerkong  have you fixed the other problem yet?

Archerkong, 2020-02-19:
@stockshuang yeah,it can work normally,and i  wrote a note，if you need i can send you

Milchdealer, 2021-01-22:
I need this too