mkolender/rtl8821CU #39: wiphy_register fail on A20 w/ 5.3.9-sunxi and 2001:331d

platima, 2020-01-27:
Hi,

I added 2001:331d to usb_intf.h as per official driver http://files.dlink.com.au/products/DWA-171/REV_C/Drivers/DWA-171_Linux_driver_v5.8.1_Driver/, and it builds and installs fine, but always seems to crash or warn, with;

Note with this DWA-171 I have to do a `usb_modeswitch -KW -v 0bda -p 1a2b` first.

I also got that official driver to compile, and have the exact same issue, so I am not 100% on the cause. Running Armbian

platima, 2020-01-27:
Note https://github.com/brektrou/rtl8821CU does not have this issue for some reason

ovhpa, 2020-02-13:
I run into the same trouble with a [driver](https://github.com/ovhpa/cudy_wu700_ac650) for cudy wu700 (ac650).

[This commit](https://github.com/ovhpa/cudy_wu700_ac650/commit/d6833430bdf4ac2b62add3fbc6af022df4267ee0) have solve it.

Probably only the changes on `rtw_cfgvendor.c` file is necessary for your issue.