mkolender/rtl8821CU #40: rPi Zero

mocallins, 2020-04-11:
Should this driver work on Buster , on a Zero, older version ?

carlosanunes, 2020-08-13:
Compilied it for RPi Zero ( CONFIG_PLATFORM_I386_PC = n, CONFIG_PLATFORM_ARM_RPI = y, CONFIG_PLATFORM_ARM_RPI3 = n ) successfully but it is not working. 

Dmesg output after installing it:
```
[   24.467167] WARNING: CPU: 0 PID: 108 at /home/pi/usbwifi/rtl8821CU/os_dep/linux/os_intfs.c:2739 rtw_os_ndevs_register+0xfc/0x140 [8821cu]
[   24.467232] Modules linked in: nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack nf_defrag_ipv4 nft_compat nft_counter nf_tables nfnetlink 8821cu(O+) brcmfmac brcmutil sha256_generic libsha256 cfg80211 rfkill raspberrypi_hwmon bcm2835_codec(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer snd videodev vc_sm_cma(C) mc uio_pdrv_genirq fixed uio ip_tables x_tables ipv6 nf_defrag_ipv6
[   24.467514] CPU: 0 PID: 108 Comm: systemd-udevd Tainted: G        WC O      5.4.51+ #1327
[   24.467526] Hardware name: BCM2835
[   24.467537] Backtrace: 
[   24.467583] [<c0014c9c>] (dump_backtrace) from [<c0014f8c>] (show_stack+0x20/0x24)
[   24.467625]  r6:00000009 r5:00000000 r4:00000000 r3:e178aaec
[   24.467670] [<c0014f6c>] (show_stack) from [<c0733b1c>] (dump_stack+0x20/0x28)
[   24.467712] [<c0733afc>] (dump_stack) from [<c0023808>] (__warn+0xdc/0x100)
[   24.467759] [<c002372c>] (__warn) from [<c00238dc>] (warn_slowpath_fmt+0xb0/0xc0)
[   24.467785]  r9:bf7d1d80 r8:00000ab3 r7:bf70ff58 r6:00000009 r5:c0a25028 r4:00000000
[   24.470124] [<c0023830>] (warn_slowpath_fmt) from [<bf70ff58>] (rtw_os_ndevs_register+0xfc/0x140 [8821cu])
[   24.470366]  r9:bf7d1dc4 r8:e024cdba r7:e024c980 r6:00000000 r5:d7d04000 r4:00000000
[   24.474985] [<bf70fe5c>] (rtw_os_ndevs_register [8821cu]) from [<bf710010>] (rtw_os_ndevs_init+0x34/0x4c [8821cu])
[   24.475067]  r10:d7d0480c r9:d9c31c78 r8:d9c90488 r7:d9c31c00 r6:e024a000 r5:d9c8dc00
[   24.475103]  r4:d7d04000 r3:00000001
[   24.479723] [<bf70ffdc>] (rtw_os_ndevs_init [8821cu]) from [<bf711598>] (rtw_drv_init+0x2a4/0x38c [8821cu])
[   24.479789]  r5:d9c8dc00 r4:d7d04000
[   24.482138] [<bf7112f4>] (rtw_drv_init [8821cu]) from [<c051e584>] (usb_probe_interface+0x148/0x26c)
[   24.482361]  r10:ffffffed r9:bf7ef0c4 r8:d9c31c00 r7:d9c8dc20 r6:d9c8dc00 r5:bf7ef1ac
[   24.482380]  r4:bf7ef0bc
[   24.482417] [<c051e43c>] (usb_probe_interface) from [<c04ad780>] (really_probe+0x214/0x358)
[   24.482463]  r10:d72d6200 r9:00000012 r8:bf7ef0c4 r7:c0b70388 r6:00000000 r5:c0b70384
[   24.482482]  r4:d9c8dc20 r3:c051e43c
[   24.482508] [<c04ad56c>] (really_probe) from [<c04adabc>] (driver_probe_device+0x6c/0x190)
[   24.482532]  r10:d72d6200 r9:bf82fee0 r8:bf7d1dfc r7:c0a25028 r6:bf7ef0c4 r5:bf7ef0c4
[   24.482547]  r4:d9c8dc20 r3:00000000
[   24.482572] [<c04ada50>] (driver_probe_device) from [<c04addd8>] (device_driver_attach+0x68/0x70)
[   24.482614]  r9:bf82fee0 r8:bf7d1dfc r7:c0a25028 r6:bf7ef0c4 r5:d9c8dc20 r4:00000000
[   24.482638] [<c04add70>] (device_driver_attach) from [<c04ade70>] (__driver_attach+0x90/0xcc)
[   24.482659]  r6:d9c8dc20 r5:bf7ef0c4 r4:00000000 r3:c051db9c
[   24.482681] [<c04adde0>] (__driver_attach) from [<c04ab6a0>] (bus_for_each_dev+0x74/0xc4)
[   24.482718]  r6:c04adde0 r5:bf7ef0c4 r4:00000000 r3:00000000
[   24.482741] [<c04ab62c>] (bus_for_each_dev) from [<c04acfc8>] (driver_attach+0x28/0x30)
[   24.482762]  r7:00000000 r6:d7e20ea0 r5:c0a99178 r4:bf7ef0c4
[   24.482783] [<c04acfa0>] (driver_attach) from [<c04ac968>] (bus_add_driver+0x138/0x1f0)
[   24.482807] [<c04ac830>] (bus_add_driver) from [<c04ae4e8>] (driver_register+0x8c/0x124)
[   24.482845]  r7:c0a99178 r6:bf7ef0c4 r5:00000000 r4:bf7ef0c4
[   24.482871] [<c04ae45c>] (driver_register) from [<c051ce38>] (usb_register_driver+0x78/0x144)
[   24.482888]  r5:00000000 r4:bf7ef090
[   24.485225] [<c051cdc0>] (usb_register_driver) from [<bf886054>] (rtw_drv_entry+0x54/0x1000 [8821cu])
[   24.485310]  r9:d72d61c0 r8:00000000 r7:c0a25028 r6:d7c22000 r5:bf886000 r4:bf7ef090
[   24.487683] [<bf886000>] (rtw_drv_entry [8821cu]) from [<c000aea4>] (do_one_initcall+0x4c/0x1c4)
[   24.487749]  r5:bf886000 r4:bf82fee0
[   24.487788] [<c000ae58>] (do_one_initcall) from [<c00959c8>] (do_init_module+0x6c/0x214)
[   24.487818]  r9:d72d61c0 r8:00000002 r7:bf82fee0 r6:c0a25028 r5:d613a0a0 r4:bf82fee0
[   24.487864] [<c009595c>] (do_init_module) from [<c0094ab8>] (load_module+0x1e64/0x229c)
[   24.487886]  r6:c0a25028 r5:00000002 r4:d7c23f28
[   24.487911] [<c0092c54>] (load_module) from [<c0095104>] (sys_finit_module+0xb8/0xcc)
[   24.487935]  r10:00000080 r9:00000000 r8:7fffffff r7:b6db58e0 r6:0000000f r5:00000000
[   24.487968]  r4:c0a25028
[   24.487993] [<c009504c>] (sys_finit_module) from [<c000918c>] (__sys_trace_return+0x0/0x14)
[   24.488010] Exception stack(0xd7c23fa8 to 0xd7c23ff0)
[   24.488032] 3fa0:                   1b44bd00 00000000 0000000f b6db58e0 00000000 b6db63f4
[   24.488054] 3fc0: 1b44bd00 00000000 00000000 0000017b 007fc778 0057b1dc 0080d4e8 00000000
[   24.488090] 3fe0: beba3180 beba3170 b6dac9d8 b6e9caf0
[   24.488116]  r9:d7c22000 r8:c00091a4 r7:0000017b r6:00000000 r5:00000000 r4:1b44bd00
```

Will try different settings, but would be good if more people could share their experiences in trying to use this with a RPi 0 ( or other ARM11 device ). Just to know if this more than just a compile issue. :/

carlosanunes, 2020-08-13:
Ok... recompiled this ( no DKMS ) and now the driver seems to be working. The warning no longer shows up in dmesg.
Having said that, I am still working on some issues...

I am using the RPi 0 W as an wireless access point ( wlan0 AP, wlan1 client). My intent is to replace the older wireless adapter I was using with a new one. After replacing it and installing the driver, I was no longer able to use wpa_supplicant. I had to edit the /etc/network/interfaces.d file ( add information for wlan1 ) and do 'ifdown' and 'ifup' to get the new adapter to be configurable by wpa_supplicant. After this, I am now able to use wpa_supplicant to connect to a network but I am no longer able to forward traffic between wlan0 and wlan1 for some reason...

All of this might not be directly related to the driver but just leaving this information to give a heads up that using an RTL8821CU adapter with a RPi 0 W is still not straightforward. Other adapters ( preferably ones with OOTB support or at least with debian driver packages available ) will be a better choice... :/

pararaum, 2020-09-01:
Same issue here, I have a PiZero V1.3 from 2015. carlosanunes, did you change anything before recompiling? I started with the "make; make install; modprobe" way...

carlosanunes, 2020-09-01:
No, did not perform any changes. Just 'sudo make' and 'sudo make install' in a fully updated system. After that, just rebooting. No errors in dmesg after that.

mocallins, 2020-09-02:
Hmmm I actually get a compile warning, and error,  and finishes with that error, step 2, I think.
Uuuugghhh been through so many truing to get these adapters to work.