mkolender/rtl8821CU #44: Does not work on Raspberry Pi OS Linux Kernel 5.4.51

AverageMaker, 2020-09-02:
As per the Raspberry Pi OS major release documentation (https://downloads.raspberrypi.org/raspios_armhf/release_notes.txt), the Raspberry Pi OS Linux kernel was changed from 4.19.118 to 5.4.51 on 20th August.

Since that time, this driver no longer works on the Raspberry Pi 4 (was working fine on the old kernel).

Is there a way we can tweak the driver to get this working on the newer kernel, or does it require a new version from Realtek?

Nodi-Rubrum, 2020-09-15:
Would like to know as well.

ovhpa, 2020-09-15:
Hello,
I have a confirmed working driver based on this one on a newer kernel (I think 5.5 the latest from Ubuntu mate).
Unfortunately it is not on Raspberry Pi (I don't own one) and I use it only for a cheap WIFI USB key ^^'
Could you perhaps try the modification in [here](https://github.com/ovhpa/cudy_wu700_ac650/commit/39d50b04536df2d58e3351f236d0ff9b7db6bd5b) and [here](https://github.com/ovhpa/cudy_wu700_ac650/commit/d6833430bdf4ac2b62add3fbc6af022df4267ee0) from my [repository](https://github.com/ovhpa/cudy_wu700_ac650) to see if you can compile and load the driver for your RPi?
I hope it can help.

Nodi-Rubrum, 2020-09-16:
Yeah, found 5.8.3 driver, and was able to compile it on x64 Debian 10, and have the dongle working on x64 Debian 10 on AMD/Intel.  But the 5.8.3 driver source does not seem to have any configuration options for Raspberry Pi.  It does have Android ARM support, so there might be a way to get something working for Pi.

I will try as you suggest in the next few days.  This last weekend, I pulled from GitHub 6 or 7 different projects that all reference the 8811 chipset, and each one fails on the newer 5.x kernels.  So basically for Raspberry Pi, every single GitHub project I tried, failed to work.  Most I could compile, but everyone generated a backtrace and tainted the kernel or such, failed.

ovhpa, 2020-09-16:
Thank you for your answer.
If I may, where did you find the 5.8.3 driver? I'm kinda lost in this 136 "forks forest" of the same driver...
Also if you encounter some trouble with the driver on Raspberry Pi, you can PM me the kernel trace or driver compilation errors, I may be able to help ;)

airdrummingfool, 2020-09-16:
In case it helps, I have successfully used [brektrou/rtl8821CU](https://github.com/brektrou/rtl8821CU) on a Raspberry Pi 3 B with Raspberry OS, kernel 5.4.51.

Nodi-Rubrum, 2020-09-19:
I got the newer driver when I got a newer dongle, the newer driver compiles with ease on Intel x86-64 Debian 10, but looking at the source code, it has no support for Pi, only Intel/AMD and some various Android OS variants... which seems odd.  Usually the Realtek website has updated stuff as well.

Oh, brektrou worked?  I thought I tried that one as well, will look at it again.

AverageMaker, 2020-09-23:
I had tried Brektrou previously without success. It may be that it doesn't work with the Pi4 for some reason?

Nodi-Rubrum, 2020-09-23:
Yeah, I could not get it to work.  I have yet to find any 8811CU variant or compatible driver to work with 5.x kernel.  Odd, because I have gotten Intel/AMD default compile to work on Intel/AMD hardware.  So this really comes down to ARM v7 support.

joaquinabian, 2020-10-01:
> In case it helps, I have successfully used [brektrou/rtl8821CU](https://github.com/brektrou/rtl8821CU) on a Raspberry Pi 3 B with Raspberry OS, kernel 5.4.51.

I can confirm the driver works fine on Raspi 4 with a PiHUT Raspberry Pi Dual-Band 5GHz/2.4GHZ USB WiFi Adapter with Antenna (Chipset rtl8821cu)

bfg100k, 2020-10-22:
> In case it helps, I have successfully used [brektrou/rtl8821CU](https://github.com/brektrou/rtl8821CU) on a Raspberry Pi 3 B with Raspberry OS, kernel 5.4.51.

To compile the driver from [brektrou/rtl8821CU], did you have to change the platform option in the MakeFile to
CONFIG_PLATFORM_ARM_RPI = y

or did you use CONFIG_PLATFORM_ARM64_RPI ?

airdrummingfool, 2020-10-22:
I should have mentioned that I built from the `5.8.1` branch.

I didn't make either of those changes (there are no `RPI`-specific config entries in the 5.8.1 makefile), but I did note that I had to do the "ARM architecture tweak" [noted in the README](https://github.com/brektrou/rtl8821CU#arm-architecture-tweak-for-this-driver-this-solves-compilation-problem-of-this-driver).

Here's the 32-bit version of the tweak:
```
sudo cp /lib/modules/$(uname -r)/build/arch/arm/Makefile /lib/modules/$(uname -r)/build/arch/arm/Makefile.$(date +%Y%m%d%H%M)
sudo sed -i 's/-msoft-float//' /lib/modules/$(uname -r)/build/arch/arm/Makefile
sudo ln -s /lib/modules/$(uname -r)/build/arch/arm /lib/modules/$(uname -r)/build/arch/armv7l
```

bfg100k, 2020-10-22:
ah ic. Well, I made the ARM tweaks and got it working. Thanks.

vogeljo, 2020-10-31:
Hey guys, still no luck on my side, even after trying all kinds of fixes. Using Arch on ARM and Raspberry Pi OS on a RPi2.
Installation works without a problem, but as soon as i attach the WiFi dongle, this appears in kernel logs and no interface is created:

```
[ 3680.263472] ------------[ cut here ]------------
[ 3680.264463] WARNING: CPU: 0 PID: 7663 at /home/pi/rtl8821cu/os_dep/linux/os_intfs.c:2739 rtw_os_ndevs_register+0x108/0x138 [8821cu]
[ 3680.264525] Modules linked in: 8821cu(O) fuse sha256_generic libsha256 cfg80211 rfkill 8021q garp stp llc joydev evdev bcm2835_codec(C) raspberrypi_hwmon v4l2_mem2mem bcm2835_isp(C) bcm2835_v4l2(C) videobuf2_vmalloc bcm2835_mmal_vchiq(C) snd_bcm2835(C) videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer snd videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 nf_defrag_ipv6
[ 3680.264711] CPU: 0 PID: 7663 Comm: kworker/0:0 Tainted: G        WC O      5.4.51-v7+ #1333
[ 3680.264716] Hardware name: BCM2835
[ 3680.264736] Workqueue: usb_hub_wq hub_event
[ 3680.264743] Backtrace: 
[ 3680.264761] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
[ 3680.264772]  r6:92a90000 r5:00000000 r4:80d95cb8 r3:36ec299d
[ 3680.264786] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
[ 3680.264799] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
[ 3680.264810]  r8:00000ab3 r7:00000009 r6:7f444d18 r5:00000000 r4:00000000 r3:36ec299d
[ 3680.264822] [<8011ff40>] (__warn) from [<801200fc>] (warn_slowpath_fmt+0xb8/0xc0)
[ 3680.264833]  r9:7f444d18 r8:00000ab3 r7:7f37e310 r6:00000009 r5:00000000 r4:80d04f48
[ 3680.265440] [<80120048>] (warn_slowpath_fmt) from [<7f37e310>] (rtw_os_ndevs_register+0x108/0x138 [8821cu])
[ 3680.265454]  r9:7f444d54 r8:bb885000 r7:7f444d18 r6:00000000 r5:9713a000 r4:00000000
[ 3680.266287] [<7f37e208>] (rtw_os_ndevs_register [8821cu]) from [<7f37e3b4>] (rtw_os_ndevs_init+0x34/0x4c [8821cu])
[ 3680.266301]  r10:9713a008 r9:9713a854 r8:b5720878 r7:b5720800 r6:bb885000 r5:b4f18a00
[ 3680.266306]  r4:9713a000 r3:00000001
[ 3680.267061] [<7f37e380>] (rtw_os_ndevs_init [8821cu]) from [<7f37f8e4>] (rtw_drv_init+0x2c0/0x38c [8821cu])
[ 3680.267069]  r5:b4f18a00 r4:9713a000
[ 3680.267450] [<7f37f624>] (rtw_drv_init [8821cu]) from [<806a7244>] (usb_probe_interface+0x108/0x248)
[ 3680.267461]  r10:ffffffed r9:7f4621b0 r8:b4f18a00 r7:7f4620c8 r6:b5720800 r5:00000000
[ 3680.267466]  r4:b4f18a20
[ 3680.267482] [<806a713c>] (usb_probe_interface) from [<80627ba4>] (really_probe+0x1ec/0x360)
[ 3680.267493]  r10:0000001b r9:7f4620c8 r8:00000001 r7:80e4f1e8 r6:00000000 r5:b4f18a20
[ 3680.267499]  r4:80e4f1e4 r3:806a713c
[ 3680.267513] [<806279b8>] (really_probe) from [<80627f14>] (driver_probe_device+0x70/0x1ac)
[ 3680.267524]  r10:b4166a50 r9:80d74584 r8:00000001 r7:80628118 r6:b4f18a20 r5:7f4620c8
[ 3680.267530]  r4:b4f18a20 r3:00000000
[ 3680.267543] [<80627ea4>] (driver_probe_device) from [<806281b0>] (__device_attach_driver+0x98/0xa8)
[ 3680.267554]  r9:80d74584 r8:00000001 r7:80628118 r6:b4f18a20 r5:92a91b5c r4:7f4620c8
[ 3680.267569] [<80628118>] (__device_attach_driver) from [<80625a24>] (bus_for_each_drv+0x88/0xcc)
[ 3680.267577]  r6:80d04f48 r5:92a91b5c r4:00000000 r3:00000060
[ 3680.267590] [<8062599c>] (bus_for_each_drv) from [<80627930>] (__device_attach+0xe0/0x148)
[ 3680.267598]  r7:b5720878 r6:b4f18a64 r5:80d04f48 r4:b4f18a20
[ 3680.267612] [<80627850>] (__device_attach) from [<806281dc>] (device_initial_probe+0x1c/0x20)
[ 3680.267622]  r8:00000000 r7:b5720878 r6:80d7459c r5:b4f18a20 r4:b4f18a20
[ 3680.267636] [<806281c0>] (device_initial_probe) from [<80626b38>] (bus_probe_device+0x94/0x9c)
[ 3680.267651] [<80626aa4>] (bus_probe_device) from [<806240fc>] (device_add+0x450/0x5f0)
[ 3680.267659]  r6:80d04f48 r5:00000000 r4:b4f18a20 r3:60000013
[ 3680.267674] [<80623cac>] (device_add) from [<806a4dd4>] (usb_set_configuration+0x4b4/0xa48)
[ 3680.267685]  r9:00000001 r8:b5720800 r7:b4166a4c r6:b4166a50 r5:b4166a50 r4:b4f18a00
[ 3680.267699] [<806a4920>] (usb_set_configuration) from [<806b0d7c>] (generic_probe+0x3c/0x88)
[ 3680.267710]  r10:0000001b r9:80d74d7c r8:00000001 r7:80e4f1e8 r6:00000000 r5:00000001
[ 3680.267714]  r4:b5720800
[ 3680.267726] [<806b0d40>] (generic_probe) from [<806a7124>] (usb_probe_device+0x34/0x4c)
[ 3680.267732]  r5:b5720800 r4:80d74d7c
[ 3680.267744] [<806a70f0>] (usb_probe_device) from [<80627ba4>] (really_probe+0x1ec/0x360)
[ 3680.267751]  r5:b5720878 r4:80e4f1e4
[ 3680.267764] [<806279b8>] (really_probe) from [<80627f14>] (driver_probe_device+0x70/0x1ac)
[ 3680.267775]  r10:80e4f7a8 r9:80d74460 r8:00000001 r7:80628118 r6:b5720878 r5:80d74d7c
[ 3680.267781]  r4:b5720878 r3:00000000
[ 3680.267795] [<80627ea4>] (driver_probe_device) from [<806281b0>] (__device_attach_driver+0x98/0xa8)
[ 3680.267806]  r9:80d74460 r8:00000001 r7:80628118 r6:b5720878 r5:92a91d7c r4:80d74d7c
[ 3680.267819] [<80628118>] (__device_attach_driver) from [<80625a24>] (bus_for_each_drv+0x88/0xcc)
[ 3680.267828]  r6:80d04f48 r5:92a91d7c r4:00000000 r3:00000058
[ 3680.267840] [<8062599c>] (bus_for_each_drv) from [<80627930>] (__device_attach+0xe0/0x148)
[ 3680.267849]  r7:b5e25478 r6:b57208bc r5:80d04f48 r4:b5720878
[ 3680.267862] [<80627850>] (__device_attach) from [<806281dc>] (device_initial_probe+0x1c/0x20)
[ 3680.267872]  r8:00000000 r7:b5e25478 r6:80d7459c r5:b5720878 r4:b5720878
[... cut ...]
[ 3680.268061] 1fa0:                                     00000000 00000000 00000000 00000000
[ 3680.268071] 1fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[ 3680.268079] 1fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[ 3680.268089]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80142df4
[ 3680.268095]  r4:ab3f6500 r3:80104648
[ 3680.268103] ---[ end trace e1c04777b4299256 ]---
[ 3680.363478] usb 1-1.2: reset high-speed USB device number 7 using dwc_otg
```

Any ideas on how to further debug?

Nodi-Rubrum, 2020-11-01:
I got a different variant of this driver to work on my Pi3.  So that might provide a clue for how to get this driver version to work.  Check out...  https://github.com/brektrou/rtl8821CU

I had to set the Makefile for ARM v7 (32 bit not 64 bit).  And follow the ARM processor tweak in the readme.  I used the DKMS based method.  It took a while to compile but it was successful.

If you check the issue list, I documented the kernel version I was running.